Z-Wave-Me/Z-Uno-Core #42: change ZUNO_SETUP_SLEEPING_MODE need hard reset of device

michapr, 2015-09-09:
if changing ZUNO_SETUP_SLEEPING_MODE mode in code you must hard reset the device before
(same as for the ZUNO_SETUP_FREQUENCY)

PoltoS, 2015-09-09:
By design. This is required by Z-Wave specs.

Lera, please add to the reference

michapr, 2015-09-10:
I'm not speaking about change the mode within the sketch, sorry for misunderstanding.
Of course this should not be changed within one sketch.

What I mean:
If device is in any sleeping mode - set by sketch 1 - and you make a new sketch with an other sleeping mode, send the compiled sketch to device - sleeping mode will not be overwritten.

This is same as for ZUNO_SETUP_FREQUENCY.
If old sketch was created for ZUNO_FREQ_RU - you can not make a new sketch for ZUNO_FREQ_EU without hard reset.

The current "warm" restart of the device seems to no be enough.

PoltoS, 2016-04-04:
Mentioned in the doc

closed 2016-04-04 by PoltoS