Z-Wave-Me/Z-Uno-Core #43: ZUNO_SETUP_SLEEPING_MODE(ZUNO_SLEEPING_MODE_SLEEPING)

michapr, 2015-09-09:
if device is in "deep sleeping" mode (expert view) - the processor is still running - executing any commands in loop().
(tried with blinking LED while receiving data via serial1)
Not sure - is this ok, or not..

PoltoS, 2015-09-09:
Also by design. You need to explicitly send the device into sleep mode.

Lera, please document it with a link to corresponding function

michapr, 2015-09-10:
I have send the device into sleeping mode - it is sleeping.

[2015-09-10 07:23:01.806] [D] [zway] Running wakeup handler for node 50
[2015-09-10 07:23:01.808] [D] [zway] SETDATA devices.50.instances.0.commandClasses.132.data.lastSleep = 1441862581 (0x55f113b5)
[2015-09-10 07:23:01.809] [I] [zway] Node 50:0 CC Wakeup: Send node to sleep
[2015-09-10 07:23:01.811] [I] [zway] Adding job: Wakeup Sleep
[2015-09-10 07:23:01.865] [D] [zway] SENDING (cb 0x15): ( 01 09 00 13 32 02 84 08 05 15 49 )
[2015-09-10 07:23:01.868] [D] [zway] RECEIVED ACK
[2015-09-10 07:23:01.871] [D] [zway] RECEIVED: ( 01 04 01 13 01 E8 )
[2015-09-10 07:23:01.872] [D] [zway] SENT ACK
[2015-09-10 07:23:01.888] [D] [zway] Delivered to Z-Wave stack
[2015-09-10 07:23:01.890] [D] [zway] RECEIVED: ( 01 05 00 13 15 00 FC )
[2015-09-10 07:23:01.891] [D] [zway] SENT ACK
[2015-09-10 07:23:01.893] [I] [zway] Job 0x13 (Wakeup Sleep): Delivered
[2015-09-10 07:23:01.894] [D] [zway] SETDATA devices.50.data.lastPacketInfo.delivered = True
[2015-09-10 07:23:01.896] [D] [zway] SETDATA devices.50.data.lastPacketInfo.packetLength = 5 (0x00000005)
[2015-09-10 07:23:01.897] [D] [zway] SETDATA devices.50.data.lastPacketInfo.deliveryTime = 4 (0x00000004)
[2015-09-10 07:23:01.900] [D] [zway] SETDATA devices.50.data.lastPacketInfo = **********
[2015-09-10 07:23:01.902] [D] [zway] SendData Response with callback 0x15 received: received by recipient
[2015-09-10 07:23:01.903] [D] [zway] SETDATA devices.50.data.lastSend = 48946375 (0x02eadcc7)
[2015-09-10 07:23:01.905] [D] [zway] Job 0x13 (Wakeup Sleep): success
[2015-09-10 07:23:01.906] [D] [zway] SETDATA devices.50.data.isAwake = False
[2015-09-10 07:23:01.907] [D] [zway] Sending the queue for node 50 into sleep
[2015-09-10 07:23:01.908] [I] [zway] Removing job: Wakeup Sleep

But anyway the processor is running /serial1 active - and LED blinking. 
Z-Wave commands (switch) will not be executed - only after wakeup (so device is sleeping - it is ok)