Z-Wave-Me/Z-Uno-Core #45: reconnect after power fail

michapr, 2015-09-10:
Please check how device is working after power fail.
If device is USB powered - (mutisensor or binary switch) - after removing power source and plug in again device will not responding to commands and will not send values from analog input.
Queue said that delivered - in log is seen that no valid value.

After uploading sketch before all was working as expected.

michapr, 2015-09-11:
maybe it is not a general problem - made today second installation at other Pi (Pi2) only with this device (to get clear logs) - here all is ok.
First Pi had sometimes low free memory and was busy (load >1.5) while inclusion process.
So maybe that's why was not full "reincluded" after power on process. Anyway, no error message was shown - so not sure...

NilsBohr, 2016-06-29:
Please check if problem still exists on the latest build

michapr, 2016-07-23:
seems to work now, can not reproduce more.

closed 2016-07-23 by michapr