Z-Wave-Me/Z-Uno-Core #47: When device is waking up - it seems not have enough time for all tasks

PoltoS, 2015-09-15:
It is going sleep before executing all tasks

void loop() {
  func_ml3();
  func_bin();
  func_ml();
  func_ml2();
  // FOR FLIRS and sleeping - enable this:
  zunoSendDeviceToSleep();
  
}

func_ml2 will not be executed.

if I will change it to
void loop() {
  func_ml2();
  func_bin();
  func_ml();
  func_ml3();
  // FOR FLIRS and sleeping - enable this:
  zunoSendDeviceToSleep();
  
}

func_ml3 will not be executed

it is easy function from examples:
void func_ml2(){
  byte currentValue = (byte) (analogRead(2) / 8);
  if ((currentValue - lastValue2 > 4) || (lastValue2 - currentValue > 4)) {
    lastValue2 = currentValue;
    zunoSendReport(ZUNO_CHANNEL_NUMBER_2);  
  }
}

michapr, 2015-09-15:
reference file for testing (sequence in loop is original - not current):
http://www.zwave-review.com/files/MultiFunctionSensor.ino

Maybe something is wrong inside?

in log you can see that sleeping init is before receiving data, not clear (for me) why:

[2015-09-15 14:27:23.335] [D] [zway] RECEIVED: ( 01 09 00 04 00 09 03 80 03 64 1F )
[2015-09-15 14:27:23.336] [D] [zway] SENT ACK
[2015-09-15 14:27:23.336] [D] [zway] SETDATA devices.9.data.lastReceived = 0 (0x00000000)
[2015-09-15 14:27:23.336] [D] [zway] SETDATA devices.9.instances.0.commandClasses.128.data.history.100 = 1442320043 (0x55f80eab)
[2015-09-15 14:27:23.336] [D] [zway] SETDATA devices.9.instances.0.commandClasses.128.data.last = 100 (0x00000064)
[2015-09-15 14:27:23.355] [D] [zway] RECEIVED: ( 01 08 00 04 00 09 02 84 07 7B )
[2015-09-15 14:27:23.356] [D] [zway] SENT ACK
[2015-09-15 14:27:23.356] [D] [zway] SETDATA devices.9.data.lastReceived = 0 (0x00000000)
[2015-09-15 14:27:23.356] [I] [zway] Node 9:0 CC Wakeup: Wakeup notification
[2015-09-15 14:27:23.356] [D] [zway] SETDATA devices.9.instances.0.commandClasses.132.data.lastWakeup = 1442320043 (0x55f80eab)
[2015-09-15 14:27:23.356] [D] [zway] SETDATA devices.9.data.isAwake = True
[2015-09-15 14:27:23.356] [D] [zway] SETDATA devices.9.data.lastSend = 2397208 (0x00249418)
[2015-09-15 14:27:23.356] [D] [zway] SETDATA devices.9.data.lastNonceGet = 2396208 (0x00249030)
[2015-09-15 14:27:23.357] [D] [zway] Running wakeup handler for node 9
[2015-09-15 14:27:23.357] [I] [zway] Job 0x13 (Wakeup Sleep): Cancelling job: Wakeup Sleep
[2015-09-15 14:27:23.357] [I] [zway] Removing job: Wakeup Sleep
[2015-09-15 14:27:23.357] [D] [zway] SETDATA devices.9.instances.0.commandClasses.132.data.lastSleep = 1442320043 (0x55f80eab)
[2015-09-15 14:27:23.357] [I] [zway] Node 9:0 CC Wakeup: Send node to sleep
[2015-09-15 14:27:23.358] [I] [zway] Adding job: Wakeup Sleep
[2015-09-15 14:27:23.375] [D] [zway] RECEIVED: ( 01 0F 00 04 00 09 09 60 0D 03 00 31 05 01 01 09 A7 )
[2015-09-15 14:27:23.376] [D] [zway] SENT ACK
[2015-09-15 14:27:23.376] [D] [zway] SETDATA devices.9.data.lastReceived = 0 (0x00000000)
[2015-09-15 14:27:23.376] [D] [zway] SETDATA devices.9.instances.3.commandClasses.49.data.1.deviceScale = 0 (0x00000000)
[2015-09-15 14:27:23.377] [D] [zway] SETDATA devices.9.instances.3.commandClasses.49.data.1.scale = 0 (0x00000000)
[2015-09-15 14:27:23.377] [D] [zway] SETDATA devices.9.instances.3.commandClasses.49.data.1.val = 9.000000
[2015-09-15 14:27:23.379] [D] [zway] SETDATA devices.9.instances.3.commandClasses.49.data.1.scaleString = "°C"
[2015-09-15 14:27:23.379] [D] [zway] SETDATA devices.9.instances.3.commandClasses.49.data.1 = Empty
[2015-09-15 14:27:23.426] [D] [zway] SENDING (cb 0x70): ( 01 09 00 13 09 02 84 08 05 70 17 )
[2015-09-15 14:27:23.429] [D] [zway] RECEIVED ACK
[2015-09-15 14:27:23.432] [D] [zway] RECEIVED: ( 01 04 01 13 01 E8 )
[2015-09-15 14:27:23.433] [D] [zway] SENT ACK
[2015-09-15 14:27:23.433] [D] [zway] Delivered to Z-Wave stack
[2015-09-15 14:27:23.447] [D] [zway] RECEIVED: ( 01 05 00 13 70 00 99 )
[2015-09-15 14:27:23.448] [D] [zway] SENT ACK
[2015-09-15 14:27:23.449] [I] [zway] Job 0x13 (Wakeup Sleep): Delivered
[2015-09-15 14:27:23.450] [D] [zway] SETDATA devices.9.data.lastPacketInfo.delivered = True
[2015-09-15 14:27:23.450] [D] [zway] SETDATA devices.9.data.lastPacketInfo.packetLength = 5 (0x00000005)
[2015-09-15 14:27:23.451] [D] [zway] SETDATA devices.9.data.lastPacketInfo.deliveryTime = 15 (0x0000000f)
[2015-09-15 14:27:23.451] [D] [zway] SETDATA devices.9.data.lastPacketInfo = **********
[2015-09-15 14:27:23.452] [D] [zway] SendData Response with callback 0x70 received: received by recipient
[2015-09-15 14:27:23.452] [D] [zway] SETDATA devices.9.data.lastSend = 2397216 (0x00249420)
[2015-09-15 14:27:23.453] [D] [zway] Job 0x13 (Wakeup Sleep): success
[2015-09-15 14:27:23.453] [D] [zway] SETDATA devices.9.data.isAwake = False
[2015-09-15 14:27:23.453] [D] [zway] Sending the queue for node 9 into sleep
[2015-09-15 14:27:23.454] [I] [zway] Removing job: Wakeup Sleep
[2015-09-15 14:27:23.535] [D] [zway] RECEIVED: ( 01 0E 00 04 00 09 08 60 0D 04 00 30 03 00 01 AF )
[2015-09-15 14:27:23.535] [D] [zway] SENT ACK
[2015-09-15 14:27:23.536] [D] [zway] SETDATA devices.9.data.lastReceived = 0 (0x00000000)
[2015-09-15 14:27:23.536] [D] [zway] SETDATA devices.9.instances.4.commandClasses.48.data.1.level = False
[2015-09-15 14:27:23.537] [D] [zway] SETDATA devices.9.instances.4.commandClasses.48.data.1 = Empty
[2015-09-15 14:27:23.556] [D] [zway] RECEIVED: ( 01 0F 00 04 00 09 09 60 0D 01 00 31 05 02 01 24 8B )
[2015-09-15 14:27:23.556] [D] [zway] SENT ACK
[2015-09-15 14:27:23.557] [D] [zway] SETDATA devices.9.data.lastReceived = 0 (0x00000000)
[2015-09-15 14:27:23.557] [D] [zway] SETDATA devices.9.instances.1.commandClasses.49.data.2.deviceScale = 0 (0x00000000)
[2015-09-15 14:27:23.558] [D] [zway] SETDATA devices.9.instances.1.commandClasses.49.data.2.scale = 0 (0x00000000)
[2015-09-15 14:27:23.558] [D] [zway] SETDATA devices.9.instances.1.commandClasses.49.data.2.val = 36.000000
[2015-09-15 14:27:23.560] [D] [zway] SETDATA devices.9.instances.1.commandClasses.49.data.2.scaleString = "%"
[2015-09-15 14:27:23.561] [D] [zway] SETDATA devices.9.instances.1.commandClasses.49.data.2 = Empty
[2015-09-15 14:27:23.575] [D] [zway] RECEIVED: ( 01 0F 00 04 00 09 09 60 0D 02 00 31 05 01 01 12 BD )
[2015-09-15 14:27:23.576] [D] [zway] SENT ACK
[... cut ...]
[2015-09-15 14:27:23.578] [D] [zway] SETDATA devices.9.instances.2.commandClasses.49.data.1.scale = 0 (0x00000000)
[2015-09-15 14:27:23.579] [D] [zway] SETDATA devices.9.instances.2.commandClasses.49.data.1.val = 18.000000
[2015-09-15 14:27:23.580] [D] [zway] SETDATA devices.9.instances.2.commandClasses.49.data.1.scaleString = "°C"
[2015-09-15 14:27:23.581] [D] [zway] SETDATA devices.9.instances.2.commandClasses.49.data.1 = Empty
[2015-09-15 14:27:43.556] [D] [zway] Job 0x13: deleted from queue
[2015-09-15 14:27:43.617] [D] [zway] Job 0x13: deleted from queue

PoltoS, 2016-07-23:
is it still relevant?

closed 2016-07-23 by PoltoS