Z-Wave-Me/Z-Uno-Core #49: Device SLEEPING - switch status change while sleeping

michapr, 2015-09-15:
ADDITIONAL to #48:
(independent of this issue - there was not chaged the status)

if changig while sleeping switch state from off - on - off you see:

[2015-09-15 15:42:58.575] [D] [zway] Job 0x13 (SwitchMultilevel Set): removing duplicate
[2015-09-15 15:42:58.575] [D] [zway] Job 0x13 (SwitchMultilevel Get): removing duplicate
[2015-09-15 15:43:00.686] [D] [zway] Job 0x13 (SwitchMultilevel Set): removing duplicate
[2015-09-15 15:43:00.686] [D] [zway] Job 0x13 (SwitchMultilevel Get): removing duplicate

Not sure how device will remember about *last* (valid) state.

NilsBohr, 2015-12-24:
Unfortunatelly right now working with external FLASH M25PE40 is not yet supported. But it is in the roadmap

michapr, 2016-01-06:
Why this depends on the external flash ?

PoltoS, 2016-01-06:
Because memory is emptied while sleeping. We will soon allow writing into FLASH and also use NonZero XRAM variables to keep state persistent during sleep,

PoltoS, 2016-04-04:
Switch will not be possible for sleeping devices. Z-Wve prohibits this ;(

closed 2016-04-04 by PoltoS