Z-Wave-Me/Z-Uno-Core #50: DEVICE in FLIRS mode - connection lost

michapr, 2015-09-16:
reference file for testing 
(same, but: ZUNO_SETUP_SLEEPING_MODE(ZUNO_SLEEPING_MODE_FREQUENTLY_AWAKE);):
http://www.zwave-review.com/files/MultiFunctionSensor.ino

after some seconds device will be "dead" in GUI - in log:

[2015-09-16 08:40:26.740] [I] [core] Notification: error (connection): Connection lost to Z-Wave device ID: 10

PoltoS, 2015-09-16:
> Кому: "Z-Wave-Me/Z-Uno-Core" <Z-Uno-Core@noreply.github.com>
> Отправленные: Среда, 16 Сентябрь 2015 г 9:45:15
> Тема: [Z-Uno-Core] DEVICE in FLIRS mode - connection lost (#50)

sure you excluded-included after changing mode to FLiRS?

----- Исходное сообщение -----

michapr, 2015-09-16:
OF COURSE! And had to reset the device before. Device is shown in UI as FLIRS device.

michapr, 2015-09-16:
problem seems to exist only after changing switch level. Transmission failed to "FLIRS" device, thats why this error (I think so)

NilsBohr, 2015-12-24:
fixed in version 0.62

closed 2015-12-24 by NilsBohr