Z-Wave-Me/Z-Uno-Core #53: Z-Uno hanging ("dead") while inclusion

michapr, 2016-01-06:
I have tested several times with actual version 0.6.2 and Z-Way 2.2.0 (Popp Gateway)
Result every time the same.
Used already your sample sketch (SimpleMultiSensor) instead of my MultiChannel sketch - result: while including (interview) Z-Uno will be dead.
I have removed all other devices, made a firmware reset of controller (RaZberry Gen5) and made a Reset (Rescue mode) of Z-Uno before uploading new sketch.
Have also removed the old "ZUNO_SETUP_FREQUENCY(ZUNO_FREQ_EU);" from sample sketch - but didn't help.
 
Attached the Log after controller reset (using the SimpleMultiSensor example sketch).
[z-uno_1.zip](https://github.com/Z-Wave-Me/Z-Uno-Core/files/79813/z-uno_1.zip)

PoltoS, 2016-01-06:
It is not hanging. According to your log, Z-Uno answers only one command and then stops answering (becomes dead). Z-Way tries few times to deliver a packet and then abondon. So, it is not hanging but waiting for the device to come back.

Now question how to reproduce this problem with Z-Uno. Any howto?

michapr, 2016-01-06:
Yes, Z-Uno becomes "dead" - and is not answering more. This I mean with "hanging".
If you reset Z-Uno - it will come back for short time. So I had after a lot of tries a full interview.
But even with full interview it goes to "dead" after short time.

Reproduce?
- Upgrading to 0.6.0.2, use your sample sketch with Arduino 1.6.5, uploading.
- Use Popp Gateway with 2.2.0 release. Reset Controller to get "fresh" HomeID. 
- Include Z-Uno.

That's all ....

It was working before fine with 2.1.x (do not remember) and 0.5.1

(About "hanging Z-Way" - it is another issue, try to reproduce with full log and will report then)

michapr, 2016-01-06:
Additional Info - While Z-Uno is "dead" a new sketch can not be uploaded.
Remove from power and restore does not help (no upload possible) - Device is as COM port still available.

michapr, 2016-01-06:
Maybe problem is Z-Way version 2.2.0
I have used at other Controller the version 2.1.2-rc17, here it is working as expected.
Interview finished and values are updating, device seems not to be "dead" after some time.
If you need this log too please tell me.

PoltoS, 2016-01-06:
Please provide both logs. Looks like something beyond our code (in SDK), since problem occurs during inclusion, but we need to check this.

michapr, 2016-01-06:
Do not think that "inclusion" is the topic.
As I said, I have included Z-Uno full, with several tries (reset,...) so that it was ready and live. But after very short time it came "dead".
Maybe any timing problem?
Will try tomorrow with other controller and UZB stick.

Vallefor, 2016-01-07:
I almost sure, that the problem in analogRead. I described it here in russian http://ask.z-wave.me/questions/2971/z-uno-analogread

I also cant include device with SimpleMultiLevel sketch. Do not know why, but condition "if (currentValue != lastValue) {" always fire and Z-Uno always try to send new value to controller. 

Any other examples work as expected.

To reproduce you need to use firmware 0.6.2 and SimpleMultiLevel sketch.

PoltoS, 2016-01-07:
Никита, линк на аск, похоже, надо проверить

PoltoS, 2016-01-07:
@379820 Лера, посмотришь?

michapr, 2016-01-07:
@Vallefor -> you are right.
I remember about it while I was testing some months ago 0.5.1 - I have modified for testing the sketch because of I had same issue... But didn't remember just now about it :-( ...

I had another issue with my own multichannel test sketch and wanted to test with standard sample at first - have taken the "wrong" sample sketch for it :)

Anyway - This sample should be redone, or everybody will get the same results....

michapr, 2016-01-07:
Sorry to say it,... but this seems not to be the main problem.
Using my old test sketch (it was working fine for "not sleeping" device):
http://www.zwave-review.com/files/MultiFunctionSensor.ino

- removed the frequency setting and the sleeping setting.
- Pin6 is for now connected to GND (via 4k resistor)

Dead device while including.

As I said - in September this sketch was working fine, without any problems.

michapr, 2016-01-07:
OK - I think I found it out.....
Situation: Made a sketch with 5 channels (link above)
Device was going to dead while inclusion, but with a lot of "wakeups" finished the inclusion process.
Device was working fine, analog read was ok. 

Then have seen that there are 6 channels in interview - instead of 5. (see image)
If I change value of latest channel in UI - device was going dead (what is "clear").

My former sketch had other channel config. ....

BUT: I really (100% sure) have excluded device everytime before including with new sketch again.

So it seems that Z-Uno remember about old config and does not overwrite the Multichannel config.
After a factory reset of Z-Uno before uploading new Z-Wave config all is working as expected.

Have reproduced it with different MultiChannel configurations.

[image]

Vallefor, 2016-01-08:
Oh... I played with zuno today and understand my problem. It was a problem with pin numeration.
I really don't understand why pin 6 = analogRead(2);

If I used analogRead on pin where nothing connected - it received a random values and each time start to send new value to controller and after few seconds it hanged.

Also when I try to use Serial.println(analogReadValue); it sends some strange symbols to serial. Documentation says that it should work http://z-uno.z-wave.me/Reference/Serial/Println/ but it ded not.
Also DEC, HEX, BIN, OCT constant are undefined.

michapr, 2016-01-08:
@Vallefor - see #56 ;)  
I agree that sending too much values in loop while inclusion process is an other problem.
I had this issue before - and forgot about this already...

But this "dead" device seems to have the decribed above reason (wrong "stored" MultiChannel config)

PoltoS, 2016-07-23:
can we close this?

michapr, 2016-07-23:
not tested - will test next week.