Z-Wave-Me/Z-Uno-Core #54: ZUNO_SENSOR_MULTILEVEL_TEMPERATURE definition

michapr, 2016-01-07:
Small change:
I would change this definition in Core_arduino.h:

#define ZUNO_SENSOR_MULTILEVEL_TEMPERATURE(GETTER) 							ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_ONE_BYTE, SENSOR_MULTILEVEL_PRECISION_TWO_DECIMALS, GETTER)	

SENSOR_MULTILEVEL_PRECISION_TWO_DECIMALS is better for temperature instead of ZERO_DECIMALS

NilsBohr, 2016-01-07:
Sorry, doesn't agree, that's a reference example macro. 
Right now you can create your own macroses. Maybe in future we will add some also.

>#define ZUNO_SENSOR_MULTILEVEL_TEMPERATURE_PREC_2(GETTER) \
>ZUNO_SENSOR_MULTILEVEL (ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_ONE_BYTE, SENSOR_MULTILEVEL_PRECISION_TWO_DECIMALS, GETTER)

michapr, 2016-01-07:
Yes, I know that it is reference example - but a temperature have digits by default, even by UPnP standard ;)
That's why suggest to improve the example ;)

NilsBohr, 2016-01-21:
Well in Z-Wave you can choose. It's not mandatory to have 2 digits.

If we change the precision to 2, then we have to change size also to 2 bytes, because if 2 digits precision with only 1 byte size -> scale will be: -1,28 to +1,28 degrees, which is not usefull ;)

And if we change the size to 2, then getter function must also return "int", so I think it's too much mess.

michapr, 2016-01-21:
yes, of course (changes above) - I made it all in my local copy - and it's working ;)
I think a normal user will use 2 digits for temperature - same as he will use no digits for lux.

I only think that for a new product we could it make more easy for a user - with standard settings.

But it is your product and marketing - you decide ;) 
It were only my "5 cents" - a suggestion ;)

closed 2016-04-04 by PoltoS