Z-Wave-Me/Z-Uno-Core #63: upload new sketch - strange issue

michapr, 2016-01-23:
Maybe I do something wrong or something is missing in documentation...
Following issue:

-> Uploading a sample Z-Uno sketch. Include device.
-> Device is working fine.
-> Exclude device.

-> Uploading a new sketch **without** Z-Wave init (any sketch, or "empty" sketch).
-> Include device
**-> old device will be included** (from old sketch)

Tried the same with factory reset or disconnecting device from power before etc. but no change.

What situation can give such issue?

NilsBohr, 2016-01-25:
OK, I think I know the cause. The set up is done only after exclusion of the device. So I think you need to exclude it. Maybe we will think of something more good.

michapr, 2016-01-25:
As I wrote - **device was excluded**. 100% sure. Multiple times have tried.
Have used Z-Way 2.2.1-rc4

NilsBohr, 2016-06-29:
Can you please recheck if the issue still exists on the latest build?

PoltoS, 2016-07-23:
can we close this?

michapr, 2016-07-23:
not tested - will test next week.

michapr, 2016-07-25:
2.003 - HC2 4.090
- Added device Temperature sensor
- Excluded device temperature sensor
- Made empty sketch (empty setup, empty loop)
- Uploades
- included device
-> Device was included as "Motion sensor"

Not sure - is it ok, if user as sample forgot to setup the channels to include such a default device.
Maybe you need it, would be nice then to add this to the docukmentation.

PoltoS, 2017-03-18:
Looks in latest HC2 firmwares it is ok

closed 2017-03-18 by PoltoS