Z-Wave-Me/Z-Uno-Core #65: analogWrite()

michapr, 2016-02-09:
Can you please recheck the analogWrite function?

Sample does not work more (SimpleDimmer).
in documentation:
`  int ledPin = 14 `
`...`
`  pinMode(ledPin, OUTPUT);   // sets the pin as output`
`...`
`  analogWrite(ledPin, val / 4)`

in sample:
`#define LED_PIN 13`
`..`
`  pinMode(LED_PIN, OUTPUT); //set up LED pin as output`
`...`
`  analogWrite(1, ((long)value)*255/99)`

But both methods does not work (any of these are "wrong", I think so 
(ledPin <-> analogPin)

NilsBohr, 2016-02-09:
We have introduced special Macroses in 0.63 Version to use with analog pins they are:

A0, A1, A2, A3 for analogRead()
PWM1, PWM2, PWM3, PWM4 for analogWrite()

Please, Try using these macroses.

michapr, 2016-02-09:
Great, PWM1 is working with internal LED.
So dokumentation must be corrected - and sample code.

PoltoS, 2016-04-04:
Done

closed 2016-04-04 by PoltoS