Z-Wave-Me/Z-Uno-Core #72: Add SensML l/s scale and doc

PoltoS, 2016-06-24:


michapr, 2016-08-08:
Can you add something like this please? (for water flow meter)

`#define ZUNO_SENSOR_MULTILEVEL_TYPE_WATER_FLOW	0x38`

`//Water flow`
`#define SENSOR_MULTILEVEL_SCALE_LITRE_PER_HOUR	0x00`

`#define ZUNO_SENSOR_MULTILEVEL_WATER_FLOW(GETTER)
ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_WATER_FLOW, SENSOR_MULTILEVEL_SCALE_LITRE_PER_HOUR, SENSOR_MULTILEVEL_SIZE_TWO_BYTES, SENSOR_MULTILEVEL_PRECISION_ZERO_DECIMALS, GETTER)`

Thanks!

PoltoS, 2016-08-08:
https://github.com/Z-Wave-Me/Z-Uno-Core/commit/d4da54c1e3843694b346b20fb5003113f7a13b59

Note, I've added 1 byte type like others. You can make pull requests in future if you want.

PoltoS, 2016-08-08:
It is not in production yet, so may be I closed it too early

closed 2016-08-08 by PoltoS