Z-Wave-Me/Z-Uno-Core #73: DHTlib compile

danielboe, 2016-06-30:
Would be great, if it is possible to compile the DHTlib from arduino. I think its a verry common usage for z-uno.

http://arduino.cc/playground/Main/DHTLib

On compile the following funktions/macros are missing (with z-uno 2.02):
```
dht_sdcpp_.cpp:461:19:error:use of undeclared identifier 'digitalPinToBitMask'
dht_sdcpp_.cpp:462:20:error:use of undeclared identifier 'digitalPinToPort'
dht_sdcpp_.cpp:463:29:error:use of undeclared identifier 'portInputRegister'
dht_sdcpp_.cpp:468:5:error:use of undeclared identifier 'delayMicroseconds'
dht_sdcpp_.cpp:505:28:error:use of undeclared identifier 'min'
```

looks not like a big problem but i dont have the hardware knowlage to implement it.

NilsBohr, 2016-06-30:
Thank you. We will add this functions in 2.03.

PoltoS, 2017-03-18:
Added

closed 2017-03-18 by PoltoS