Z-Wave-Me/Z-Uno-Core #74: Serial.parseFloat() crashes

danielboe, 2016-07-04:
The parseFloat() and parseInt() function makes the zuno crash. only reset helps to make it available via serial for reprogramming.
sample code:

```
void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  Serial2.begin(9600);
  Serial.println("Init");
  Serial.println();
}

void loop() {
  // Read serial input:
  while (Serial2.available() > 0) {
    char inChar = Serial2.read();
    Serial.println(inChar);
    if (inChar == 'A') {
        for (int i = 1; i <=2; i++) {
          float test = Serial2.parseInt();
          Serial.print("float: ");
          //Serial.println(test);
        }
      Serial.println("Select A");
    }
  }
}
```

NilsBohr, 2016-07-04:
Thank you for your report.
What version are you using? 2.02?

danielboe, 2016-07-05:
Ah sorry, I forgot. Its version 2.02, yes. Also itos() is not implemented, so I actually see no way to convert string->number. Thanks

danielboe, 2016-07-05:
Here is a simple function to convert char-arrays to numbers:
```
int a2i(const char *s)
{
 int sign=1;
 if(*s == '-'){
  sign = -1;
  s++;
 }
 int num=0; //maybe double needed if greater values
 while(*s){
  num=((*s)-'0')+num*10;
  s++;   
 }
 return num*sign;
}
```

PoltoS, 2016-07-23:
Please try 2.0.3

PoltoS, 2016-08-24:
Can we close it? Does it work now on 2.0.5?

closed 2017-03-18 by PoltoS