Z-Wave-Me/Z-Uno-Core #80: update from 2.0.2 to 2.0.3 error

michapr, 2016-07-17:
Hi,

can not update from 2.0.2 to 2.0.3 - get follwing error writing boot loader:
...
Writing NVM data                         [=================>                                ] (35%)
Writing NVM data                         [=================>                                ] (35%)Traceback (most recent call last):
 
  File "<string>", line 1149, in serial_api_fwfunc
  File "<string>", line 884, in reflashFirmware
  File "<string>", line 815, in writeFileToNVM
ProgrammerError: Can't write NVM data! at=39c00

or

ProgrammerError: Can't write NVM data! at=5d800
Error  0	 Unknown exception:Can't write NVM data! at=5d800

Any idea?

Michael

michapr, 2016-07-17:
after complete reset (there was a sketch inside) firmware update was possible.
Maybe make sense to add it to troubleshooting page?

Michael

michapr, 2016-07-17:
Do not understand what you mean, sorry.

 

Have you used USB in your sketch?

PoltoS, 2016-07-17:
Your previous sketch did something that blocked Z-Uno bootlader from communications via USB. This is usually either some insane loop without delays or USB usage from sketch.

[Rescue mode](http://z-uno.z-wave.me/getting-started/troubleshooting#Z-UnoHangs) should help in both cases.

michapr, 2016-07-18:
Hi,

 

no, no USB usage in sketch.

 

Loop without delays – yes. A delay should not be mandatory in a loop, right?

There were called 2 functions in loop 

First is looking for changes button state and increase counter if state was changed

Second will send every 10 seconds (if(currentTime >= (cloopTime + 10000)) )  the report to Z-Uno.

 

Z-Uno was in ALWAYS_AWAKE mode.

 

So I really think nothing “critical” …

 

Best regards

Michael

michapr, 2016-08-22:
I think outdated. can be closed

closed 2016-08-22 by michapr