Z-Wave-Me/Z-Uno-Core #82: Add some technical parameters for Z-Uno at website

michapr, 2016-07-22:
can you please add some information about the Z-Uno on website - users could need it for right usage
- max current for the digital output pins
- max current for the 5V / 3.3V output if external power will be used (for supply of additional components)
  (I think there mybe a thermal limit)
- used current for the different modes (always on, sleeping or FLiRS)
- Limits for power supply for "3V" and "5V" - **as sample** module will work for
  3V input ( 1.8...3.3V max  (???) )
  5V input ( 4.2V ... 6V max (???) )
  -> so user could define a battery type

Thanks!

michapr, 2016-07-27:
Thanks, something already done! ;)

Can you please add this info?

- Limits for power supply for "3V" and "5V" - as sample module will work for 3V input ( 1.8...3.3V max (???) ) 5V input ( 4.2V ... 6V max (???) ) -> so user could define a battery type
- max current for the 5V / 3.3V output pin if external power will be used (for supply of additional components) (I think there mybe a thermal limit)

PoltoS, 2016-07-27:
@p0lyg0n1 подскажешь? максимум известных тебе цифр о потреблении, макс. токах и прочему

PoltoS, 2016-08-24:
Is this enough z-uno.z-wave.me/technical/?

michapr, 2016-08-25:
missing (please check, what is correct,... ;) ...):
working battery range: for 3V input: 2.7V...3.3V 
input power range for 5V input: 4V... 6V (?)  - if it could be used as input
max output current for 3.3V output (ext.devices) if supply by pin "7-16V" : 100mA (?)
max output current for 5V output (ext. devices) if supply by pin "7-16V" :  100mA (?)

*Note* Do not supply the Z-Uno with ext. Battery AND USB-power/"7-16V" input, the battery power input is not protected

maybe show at any page tis detail: "by unsoldering R12" - where R12 is located ?