Z-Wave-Me/Z-Uno-Core #84: OpenZWave timeout

michapr, 2016-07-22:
Not sure why, but can reproduce.

Simple ZUNO_SENSOR_MULTILEVEL_TEMPERATURE 
(same sketch as Simple_Multisensor - only other class)

while iclusion see timeouts:

2016-07-22 10:32:38.349 Info, Node029, Request RTT 23 Average Request RTT 26
2016-07-22 10:32:48.326 Error, Node029, ERROR: Dropping command, expected response not received after 1 attempt(s)
2016-07-22 10:32:48.329 Detail, Node029, Removing current message
2016-07-22 10:32:48.329 Detail, Node029, Notification: Notification - TimeOut
2016-07-22 10:32:48.330 Info, Node029, Sending (Query) message (Callback ID=0xc5, Expected Reply=0x04) - MultiInstanceCmd_Get (Node=29): 0x01, 0x0a,
0x00, 0x13, 0x1d, 0x03, 0x60, 0x04, 0x5e, 0x25, 0xc5, 0x22
2016-07-22 10:32:48.336 Detail, Node029,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-07-22 10:32:48.337 Detail, Node029,   ZW_SEND_DATA delivered to Z-Wave stack
2016-07-22 10:32:48.354 Detail, Node029,   Received: 0x01, 0x05, 0x00, 0x13, 0xc5, 0x00, 0x2c
2016-07-22 10:32:48.355 Detail, Node029,   ZW_SEND_DATA Request with callback ID 0xc5 received (expected 0xc5)
2016-07-22 10:32:48.355 Info, Node029, Request RTT 24 Average Request RTT 25
2016-07-22 10:32:58.332 Error, Node029, ERROR: Dropping command, expected response not received after 1 attempt(s)
2016-07-22 10:32:58.334 Detail, Node029, Removing current message
2016-07-22 10:32:58.335 Detail, Node029, Notification: Notification - TimeOut
2016-07-22 10:32:58.336 Info, Node029, Sending (Query) message (Callback ID=0xc6, Expected Reply=0x04) - MultiInstanceCmd_Get (Node=29): 0x01, 0x0a,
0x00, 0x13, 0x1d, 0x03, 0x60, 0x04, 0x60, 0x25, 0xc6, 0x1f
2016-07-22 10:32:58.342 Detail, Node029,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2016-07-22 10:32:58.342 Detail, Node029,   ZW_SEND_DATA delivered to Z-Wave stack
2016-07-22 10:32:58.360 Detail, Node029,   Received: 0x01, 0x05, 0x00, 0x13, 0xc6, 0x00, 0x2f
2016-07-22 10:32:58.360 Detail, Node029,   ZW_SEND_DATA Request with callback ID 0xc6 received (expected 0xc6)
2016-07-22 10:32:58.360 Info, Node029, Request RTT 23 Average Request RTT 24
2016-07-22 10:33:08.337 Error, Node029, ERROR: Dropping command, expected response not received after 1 attempt(s)
2016-07-22 10:33:08.340 Detail, Node029, Removing current message
2016-07-22 10:33:08.340 Detail, Node029, Notification: Notification - TimeOut
2016-07-22 10:33:08.341 Info, Node029, Sending (Query) message (Callback ID=0xc7, Expected Reply=0x04) - MultiInstanceCmd_Get (Node=29): 0x01, 0x0a,
0x00, 0x13, 0x1d, 0x03, 0x60, 0x04, 0x70, 0x25, 0xc7, 0x0e

Have reincluded device second time - then all was ok.

Remember that told about this issue already some time ago.

There is no fast loop or something like this inside - in loop() is a delay(5000) inside for testing, but I think this must not be a problem here. zunoSendReport will be called "only" every 5 seconds.
But sketch (zunoSendReport) will work only after include - right?

Configuration: Raspberry pi 3 + UZB Stick

PoltoS, 2016-07-22:
Very interesting! 0x60, 0x04, 0x5e is MultiInstance V1 Get. This MultiInstance V1 is deprecated 5 years ago and is not madatory to support in MultiChannel V3 devices, so we don't answer it.

Please make a ticket in OZW.

Colleagues says you are using very old OZW like 1.1. Should be at least 1.4

closed 2016-07-22 by PoltoS

michapr, 2016-07-22:
I use OZW Version: 1.4-77-g40eb730

Of course can open a ticket there.
Interesting for me is, that a reinclude was working - why if the reason is an outdated MultiInstance V1 ?

PoltoS, 2016-07-23:
Because it is not Z-Uno issue, but OZW issue. It is worth to reference the OZW issue here.