Z-Wave-Me/Z-Uno-Core #85: HomeCenter2 - no values

michapr, 2016-07-22:
Using HC2 4.083 beta

Have included device - without problems:
[15:05:44] Home Center ist im Lernmodus.
[15:05:48] Modul hinzugefügt
[15:05:48] Hinzufügen des Netzknoten Status Protokols abgeschlossen
[15:05:48] Hinzufügen des Modul abgeschlossen.
[15:05:52] ID 176: Set association for group 1
[15:05:52] ID 176: Received association report from group 1
[15:05:52] ID 176: Received association report from group 1
[15:05:53] ID 176: Set association for group 1
[15:05:53] ID 176: Received association report from group 1
[15:05:56] ID 176: Device configuration finished
[15:05:57] ID 176: Device configuration finished

[image]

Values will not be changed.

Sketch attached - same sketch is working with OZW (after reinclude - see #84) 
If I will enable the Serial.print - COM port maybe lost. Or nothing willbe printed out.

Using OZW - values will be printed out. No idea how this is connected...

[ZUnoSimpleMultiSensor1.zip](https://github.com/Z-Wave-Me/Z-Uno-Core/files/378257/ZUnoSimpleMultiSensor1.zip)

PoltoS, 2016-07-23:
Is it 100% reproducible? We will try it on monday.

We need Z-Uno version too (does it happen on 2.0.3 too?)

michapr, 2016-07-25:
Is working with HC2 4.090 and Beta release 2.0.3
Problem exist at stable release 2.0.2 only

PoltoS, 2016-07-25:
So we can close it? Looks we fixed it too fast ;)

michapr, 2016-07-25:
yes, for beta 2.0.3 can be closed - it is existing, while current stable release is out.

closed 2016-07-25 by michapr