Z-Wave-Me/Z-Uno-Core #86: compiler error if "wrong" filename

michapr, 2016-07-22:
Rename the Project to "Z-UnoSimpleMultiSensor_1" and save it

you will get a compiler error:

sdcc: Generating code...
at 1: warning 119: don't know what to do with file ''. file extension unsupported
_Z-UnoSimpleMultiSensor_1_ucxx.c:203: syntax error: token -> '-' ; column 23
Error. SDCC returned: 1

[ZUnoSimpleMultiSensor1.zip](https://github.com/Z-Wave-Me/Z-Uno-Core/files/378274/ZUnoSimpleMultiSensor1.zip)

Something wrong there...

PoltoS, 2016-08-24:
As I remember, - is not allowed in file names

michapr, 2016-08-25:
It seems it is allowed - every new open sketch (depending om version settings maybe) will have this format, can be saved and compiled.
sketch-jul22b.ino

I can compile sketchs with this filenames for other arduino boards (other compiler)

PoltoS, 2017-03-18:
Fixed

closed 2017-03-18 by PoltoS