Z-Wave-Me/Z-Uno-Core #93: Ultrasonic Sensor HC-SR04

michapr, 2016-08-22:
Can you provide any solution for using the Ultrasonic Sensor HC-SR04?

Because of missing miros and missing PulseIn it is not easy to use this popular sensor.

Would be nice if could be possible.

Thanks!
Michael

PoltoS, 2016-08-26:
This worked for me. The 6.9 divisor is a calibration value obtained after some tests. We will probably include this pulseIn in future release making more calibration and checks.

```
int readPin = 9;
int triggerPin = 10;

void setup() {
  Serial.begin();
  pinMode(readPin, INPUT);
  pinMode(triggerPin, OUTPUT);
  digitalWrite(triggerPin, LOW);
}

void loop() {
  digitalWrite(triggerPin, LOW);
  delayMicroseconds(2);
  digitalWrite(triggerPin, HIGH);
  delayMicroseconds(10);
  digitalWrite(triggerPin, LOW);
  Serial.println(pulseIn(readPin, 10 * 1000) / 58); // convert to cm, see datasheet
  delay(1000);
}

unsigned int pulseIn(s_pin pin, int timeoutCount) {
  const float count2us = 6.9; // calibration value
  unsigned int count;
  pinMode(pin, INPUT);
  timeoutCount = (unsigned int)((float)timeoutCount / count2us); // convert to cycles

  // wait for LOW state (previous pulse end)
  count = 0;
  while (digitalRead(pin) != LOW)
    if (count++ == timeoutCount)
      return 0;

  // wait for HIGH state (pulse start)
  count = 0;
  while (digitalRead(pin) == LOW)
    if (count++ == timeoutCount)
      return 0;

  // wait for LOW state (pulse end)
  count = 0;
  while (digitalRead(pin) != LOW)
    if (count++ == timeoutCount)
      return 0;

  return (unsigned int)((float)count * count2us); // convert from cycles
}
```

michapr, 2016-08-29:
Thanks, it is working!

Changig the timeoutCount to 30,000 I have seen valid results > 2m.
Maybe there will be needed any "fine tuning" for count2us for better results - but it is working!

Michael

michapr, 2016-10-26:
Final sketch for 2.0.6 is working, but I think he PulseIn lib should be calibrated a bit.
The values are to small, not only a little bit ;)
If calibration not in PulseIn - then maybe in sketch, so that it will work fine for users...

closed 2016-12-20 by michapr