Z-Wave-Me/Z-Uno-Core #95: DS18B20 sample 1Wire

michapr, 2016-08-25:
I'm a little bit confused about the 220 Ohm resistor.
Is it true - or a mistake?
http://z-uno.z-wave.me/examples/1-wire-ds18b20-temperature-sensor/
look here:
https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
-> 4.7 kOhm

PoltoS, 2016-08-25:
> Кому: "Z-Wave-Me/Z-Uno-Core" <Z-Uno-Core@noreply.github.com>
> Отправленные: Четверг, 25 Август 2016 г 21:36:42
> Тема: [Z-Wave-Me/Z-Uno-Core] DS18B20 sample 1Wire (#95)
> I'm a little bit confused about the 220 Ohm resistor.
> Is it true - or a mistake?
> http://z-uno.z-wave.me/examples/1-wire-ds18b20-temperature-sensor/
> look here:
> https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
> -> 4.7 kOhm

indeed recommended is 4.7k. Updated. Thanks. Though I don't think it is really important. 

PS 

----- Исходное сообщение -----

closed 2016-08-25 by PoltoS

michapr, 2016-08-26:
> Though I don't think it is really important.

it is...
"Digital pin out logical 0: < 0.4 V, max current sinking 8 mA"
-> 2,9V / 220 Ohm = 13.2 mA  really too much for such an application

Even not sure will it be fine for DS18B20 with such a current on output.
Never have used R<2kOhm for 1-Wire - even for long lines... ;)