Z-Wave-Me/Z-Uno-Core #96: DB18B20 sample return wrong values

michapr, 2016-08-26:
at first - would be nice if the funtion would allow to work with more precision as return value.
The integer value is for temperature not optimal ;)
Would like to suggest to format the value after receiving from function.

Function getTempC10 will return wrong values - it receive the right values from sensor, but conversion is wrong.

as sample:
will read: 167:1:75:70:127:255:9:16:224  (9 bytes)
temp (combined data[0] and [1] will be 423
final temp after `(temp * 50) >> 6 ;` : **330**
I will get this return.

As I wrote above, I would work with float and convert the result to full value.

michapr, 2016-08-26:
BTW: could you, if possible, rename your library in a future release to DS18B20Z as sample, so that no conflict could be with already installed lib?
Will be more transparent for user, that this is a special lib.

PoltoS, 2016-08-26:
After fixing it check that sample code on the site is still valid

michapr, 2016-10-05:
with 2.06 it was fixed - thanks.

closed 2016-10-05 by michapr