Z-Wave-Me/Z-Uno-Core #97: sleeping mode

michapr, 2016-09-18:
Please provide any sample for users for sleeping mode

As sample a temp sensor reporting every 10 minutes or/and a temp sensor reporting if values was chaged by 1°C

Reading documentation it is not clear for me how it can work ;)

Thanks!

PoltoS, 2016-09-18:
This page describes it http://z-uno.z-wave.me/Reference/zunoSendDeviceToSleep/

michapr, 2016-09-19:
> This page describes it http://z-uno.z-wave.me/Reference/zunoSendDeviceToSleep/

OK, this page is online again, and something is described here.
But anyway...
> Z-Uno will wake up after wakeup period (set up via Wakeup Command Class) 

Is this the final solution? Or will it be changed using the future config parameters?
Most devices have a setup option for (as sample) checking temperature measure interval - independent of the wakeup time of device.
Would be nice if Wakeup Command Class is configurable via Arduino code (I think it is, only do not know how ;) ....)

michapr, 2016-09-19:
Additional: Could you please publish a simple sample (DS18B20 sensor) in sleeping mode?
My first try was not successful, so I think it could be helpful for other users too ;)
(didn't got Temperature values, but "default backup value" ;) )
Thanks!

michapr, 2016-09-19:
I found out how it can work - but any small optimization in lib would be nice ;)
Even not sure - is this the correct way...

`void setup() {`
`DS18B20Sensor ds1820(&ow); `
`    Serial.begin();`
`    Serial.println("start");  `
`}`
` `
`void loop() {`
`    delay(20);`
`    // obtaining readings from the sensor ds18b20`
`    temp=getTempds18b20();`
`   // send data to channel`
`    delay(20);`
`    zunoSendReport(1);     `
`    zunoSendDeviceToSleep();`
`}`

michapr, 2016-10-11:
Sleeping mode with 18B20 is not working. Send first value and never wake up.
Yes, I have modified the wakeup time to 240seconds ;)

Normal sleeping mode (without sensor, but fixed value) is working fine.
Can you please recheck?

Additional - you are sure that you have icluded the right measure time of 18B20 in library ?
It takes up to 750ms, depending on resolution.

Attached the sketch.
(why it does not acccept *.zip?...)

[zUno_18b20_sleep.txt](https://github.com/Z-Wave-Me/Z-Uno-Core/files/522672/zUno_18b20_sleep.txt)

PoltoS, 2016-10-11:
Related to https://forum.z-wave.me/viewtopic.php?f=3427&t=23876

michapr, 2016-10-12:
will be solved (both - invalid first value and "hanging sleep" if you modify the OneWire.cpp as described in forum (link above)

Please close if modification was done.

PoltoS, 2017-03-18:
Fixed

closed 2017-03-18 by PoltoS