Z-Wave-Me/Z-Uno-Core #98: 18B20 usage

michapr, 2016-09-18:
How to request multiple 18B20 on same OneWire line?

In most cases there are multiple sensors on one line - that's why there should be a function for looking for address of the 18B20 - and using this address the sensor value will be requested.

Using 2 sensors (parallel) in this time will not work.

Maybe you can prvide a sample code using your lib?

Thanks!
Michael

p0lyg0n1, 2016-09-18:
In current version you can do it if you know uid(rom) of your sensors. You can scan its rom when sensor connect to bus alone and then use it to get its value when you have many sensors connected to the bus. We will provide full search procedure for OneWire in the next releases.

michapr, 2016-09-19:
ok, then you can close the issue - maybe add any hint in code comment section?

closed 2016-12-20 by michapr