Z-Wave-Me/Z-Uno-Core #102: Sleeping Z-Uno and DHT sensors

michapr, 2016-11-04:
Issue is not easy to describe...

Following steps:

-exlude Z-Uno (to be sure...)
-Upload this sketch without the delays in line 28,36
-Device will not go into sleep mode (LED is blinking)
-including the device in Z-Way -> ZWay show sleeping device. Changing wake-up, all seems to be ok, but device will never wakeup and change values. LED is still blinking

-exlude Z-Uno 
-Upload this sketch **with** the delays in line 28,36
   ->LED will (in most cases) still blink - Z-Uno is not sleeping
-Make factory reset - upload same sketch **with** the delays in line 28,36 again
   -> Z-Uno will go into sleep mode
-including the device in Z-Way, set wake-up - all seems to work fine, values will be changed after wake-up.

-edit sketch - as sample change delay from 2000 to 3000 seconds, upload sketch (rescue mode)
  -> Z-uno will not go into sleep more
  -> exclude, factory reset, upload - Z-Uno will go into sleep

If Z-uno will go into sleep mode, you can not upload changed sketch more (rescue mode of course) - Z-Uno will not go into sleep more, only after factory reset.

Strange issue, I hope you understand. It is reproducable.
[ZUno2DHT22a.zip](https://github.com/Z-Wave-Me/Z-Uno-Core/files/570799/ZUno2DHT22a.zip)

First issue seems to be that DHT sensors are not ready. They need 2seconds (datasheet), but in lib I have seen lower value. Changing this only didn't help.

Without DHT sensor lib (with fixed values) all is working as expected

Michael

michapr, 2016-11-05:
Maybe found first issue in DHT lib:

in ZUNO_DHT.cpp:

`void DHT::begin(void) {`
`  // set up the pins!`
`  pinMode(_pin, INPUT_PULLUP);`
`  _lastreadtime = 0;`
`}`

later:

`  if (!force && ((currenttime - _lastreadtime) < 2000)) {`
`    return DHT_RESULT_PREVIOUS; // return last correct measurement`
`  }`

After wakeup the _lastreadtime will be 0 - so no value will be reported.
I think initial value should be 2000

`void DHT::begin(void) {`
`  // set up the pins!`
`  pinMode(_pin, INPUT_PULLUP);`
`  _lastreadtime = 2000;`
`}`

Can you recheck it?

PoltoS, 2016-11-05:
@p0lyg0n1 we should use NZ for such values.

michapr, 2016-11-05:
@PoltoS I think in this case it is not needed - for sleeping the sleeping time > 2seconds, so this value can be ignored (if calling function once).
If not calling once or not sleeping - it will work, I think so. Only initial value for first call should be 2000, so that will be executed.

michapr, 2016-11-05:
My solution is not full correct - because of currenttime will be some ms only... So could be better to check for (initial value)  ( _lasttime =0  ||  (currenttime - _lastreadtime) < 2000)) or something like that.
But there is something else wrong - even if it is working - Z-Uno will hang after some time (maybe hours) ...