Z-Wave-Me/Z-Uno-Core #115: ZUNO C++ class inheritance with (pure) virtual functions as "interface" / "polymorphysm"

charly61ch, 2018-04-12:
Dears,
First, CONGRATULATIONS for the Z-UNO product. 
Fantastic, simple, genial !!!!! 
**BRAVO LES RUSSES !**

I however have a compilation problem when I try to pass, as parameter, a pointer to an object which inherits a class, as "interface", for later callback usage.

**Problem Description :**
I have a class C_I in which I just declare a virtual function. (interface)
I have a class C_C which inherits C_I and implements the functions.
I have a class C_U which must call a function of C_C (virtually defined in C_I)
In C_U, I have a function "setInterface(C_I * p)" which must be called by a "client" to initialize the callback object pointer.

In my main.ino  main project module :
I declare an object O_C of class C_C, and an object O_U of class C_U, both statically instantiated.
In the "Setup" function, I call the "O_U.setInterface(&O_C)",...
... and that, I cannot compile !
I tried a lot of possibilities, declaring an intermediate pointer, inside the "setup" function, outside the function; I tried to instantiate my object inside, outside the Setup function..... chess, chess, chess.

The compiler error is :
/tmp/arduino_build_778464/cObjSibir_ucxx.c:491: error 101: too many parameters 

The generated intermediate code is :
(which seems however not too bad)
//class cObjSibir IMPLEMENTATION
//--------------------------------------------------------------------------------
void * ​__cxx__vtable__cObjSibir[2];
void __cxx__cObjSibir__init00(__xdata void * v__this)
{
     __cxx__iObjIface__init00(v__this);
     ((cxx__class__cObjSibir*)v__this)->__cxx__vmethod__table = ​__cxx__vtable__cObjSibir;
...
}

Attached, you will find fragments of my original code, and the intermediate c file.

Hoping that my explanation is clear, and somebody can answer,....
... first if I have the right doing so, 
... and eventually help me to solve this issue.

Thank's,
Regards,
Charly C.

[zuno_virtual_inherit_problem.txt](https://github.com/Z-Wave-Me/Z-Uno-Core/files/1904697/zuno_virtual_inherit_problem.txt)

[cCommon_ucxx.c.txt](https://github.com/Z-Wave-Me/Z-Uno-Core/files/1904714/cCommon_ucxx.c.txt)

charly61ch, 2018-04-13:
Dears,

I would like to use my favorite OOP construction like "Interface Injection".
I have a class cIF, which declares a pure virtual function, as an interface, like a callback.
I have a class cOBJECT, which inherits and implements the function.
I have a class cHANDLE, which don't know cOBJECT, but knows cIF. This class has a function setInterface(cIF * pIf), which stores a "client" interface pointer object.
In my "main.ino" module, I instantiate statically a cOBJECT obj and a cHANDLE hdl.
In the Setup function, I would like to call hdl.setInterface(&obj).

BUT I cannot compile this last statement.
I cannot cast in any other way, any pointers.
The error is : "indirections to different types assignment".
from type 'struct cxx__class__cOBJECT_s generic* xdata'
to type 'struct cxx__class__cIF_s generic* auto'
I understand the error, ant the difficulty for the compiler to handle the memory spaces, and my question is :
Is my wish possible ?
Can I expect a compiler upgrade to support that ?

Thanks in advance for an answer,
Regards,
Charly C.

p0lyg0n1, 2018-04-13:
Hi, Charly! Please provide your test code in one module (lets call it main.ino). I'll try to get it work. If it's not a big deal I'll add this to next release of compiler. Architecture of the chip is very ancient(Too old to rock&roll :) ). It has only about 100 bytes of stack for all, but I'll try to take it to its limits.

charly61ch, 2018-04-13:
Hello @p0lyg0n1 
Thanks for your answer.
Here joint the "whole" :-) code I would like to use as "basis".
(little "academic" for now, but will grow :-) )

`iObjIface` defines the interface
`cObjSwitch` has to callback `client->SetValue`. The client is registrable by calling its `setObjIface` function
`cObjSibir` is the "client" object, which inherits iObjIface,

SIB010.INO :
```
cObjSibir   Sibir; // inherits iObjIface
cObjSwitch  Switch;
void setup()
{
   Switch.setObjIface(&Sibir); // Not compilable
also :
   iObjIface * p = &Sibir; // Not compilable
   iObjIface * p = (iObjIface*)&Sibir; // Not understandable by compiler (but not dramatic if implicit cast will work)
}
```
So given, the code compiles, except SIB010.ino, line 33

Still thanks,... and more if you have a solution. :-)
And very big congratulation to guys who developed a so simple, easy and genial customizable zwave device.

[Ptr_To_Inherited_Base_Class Cannot_UpCast.zip](https://github.com/Z-Wave-Me/Z-Uno-Core/files/1909616/Ptr_To_Inherited_Base_Class.Cannot_UpCast.zip)

Regards,
Charly