Z-Wave-Me/Z-Uno-Core #118: How to implement dynamic channels in 2.1.5? zunoCallback and callback_data seem to have changed

copumpkin, 2019-07-21:
I'm using 2.1.5 and am just attempting to compile locally on macOS right now, so my board setup shouldn't matter much.

I'm trying to use some of the dynamic channel examples like [this one](https://z-uno.z-wave.me/examples/multiple-dynamic-1-wire-ds18b20-temperature-sensors/) but it seems like the struct definition for `callback_data` has changed. First of all, it's no longer a pointer, so the dereferencing `->` doesn't work. But even if I change that to a `.`, the struct no longer has a `param` field, and instead seems to have `param1`, `param2`, and `result`. The examples haven't been updated so I'm not sure what the correct way to do dynamic channels is anymore.

Also, if I even include a definition of `zunoCallback` (e.g., a no-op `void zunoCallback() {}`) in my code, compilation seems to fail with a mysterious Python error like this:

```
Preprocessing file: Custom.c with SDCPP... 
Traceback (most recent call last):
  File "/private/tmp/zuno-packager/build/zuno_toolchain-00.08.80-macosx/build/compiler/out00-PYZ.pyz/uCxx", line 7842, in compileArduinoSketch
  File "/private/tmp/zuno-packager/build/zuno_toolchain-00.08.80-macosx/build/compiler/out00-PYZ.pyz/uCxx", line 6697, in makeZWaveTemplate
ValueError: too many values to unpack
```

Simply removing the `zunoCallback` fixes that compilation error but then I'm not sure how to return my value.

sjlb197, 2019-10-19:
I'm facing this same problem - did you find a solution?  Not found anything else yet by searching, so hoping you may have figured it out!

PoltoS, 2019-10-20:
Please have a look on the documentation
https://z-uno.z-wave.me/Reference/ZUNO_START_CONFIG/

It had changed a lot since 2.1.5