Z-Wave-Me/Z-Uno-Core #120: [feature request] Support PlatformIO dev tool

ncolomer, 2019-10-05:
It would be great to code zuno firmwares using [PlatformIO](platformio.org) ecosystem.

The necessary compatibility glue was reduced to minimum and is documented [here](https://docs.platformio.org/en/latest/platforms/custom_platform_and_board.html).

Since zuno is built on top of custom zwave microcontroller, It would probably require declaring both new platform and new board.

wdyt?

BugHunterPhilosopher, 2019-12-04:
+1 for me! That would be great!

PoltoS, 2019-12-06:
What is the advantage over using the current Arduino IDE? The compiler can even now be executed as a standalone CLI.

The code generation can be done by you like we do it for the Z-Uno Shield: https://z-uno.z-wave.me/shield/configurator/

Please describe your idea in more details

BugHunterPhilosopher, 2019-12-06:
For me, that's the very difference between a code editor and a full-IDE: with Platform.io you can easily unit test, debug, CI, remote OTA, etc. With just the plain-vanilla Arduino IDE some of these things are hard, some others are impossible.

PoltoS, 2019-12-06:
Our compiler and programmer is available as CLI, so you can easily integrate it with Platform.io.

Please check this page: https://z-uno.z-wave.me/tools/

toepi, 2020-01-12:
Are you willing to help porting to platformio? I've try to create a new platform, but fail at download toolchain, because our TLS is deprecated. So please update our TLS setting and put a package.json inside to archive.

Another issue that I've you have provide a linke to use your tools at command line but it not explain how to compile a project - and I fail also at this stage, `compile build mystrecht.ino` doesn't work.