Z-Wave-Me/Z-Uno-Core #121: Temperature reported 288.42 and 197.08 with 1-wire and freq awake

silverman10, 2020-01-23:
_Please fill your bug report here_
I am using almost similar sketch as in the example, but with sleeping mode frequently awake, with BeamWU at 30.
Its working most of the time, but i get temperature readings of 288.42 and 197.08 on the zwave channel. I have tried to add a delay of 2000 in the loop after eack wake up, but no avail.

**More info:**
- **Z-Uno bootloader version:** 2.1.5 (release version or test-ucxx?)
- **Security:** none
- **Frequency:** EU
- **Device included:** yes
- **Device included securely by controller:** no
- **Controller:** other (openhab)

**Connected peripherals:**
_Please add here a description, photo and/or schematics of your peripherals_

**Code:**
_If you can minimize your code to still reproduce the bug, please do it before reporting_
```C
#include "ZUNO_DS18B20.h"

#define DS18B20_BUS_PIN 11                  // Pin to which DS18B20 bus is connected
#define N_SENSOR 2                          // Number of DS18B20 sensors
// You need to adopt ZUNO_SETUP_CHANNELS and add more getters if tou need more sensors

OneWire ow(DS18B20_BUS_PIN);
DS18B20Sensor ds18b20(&ow);

#define ADDR_SIZE 8                         // Size of address of devices on 1-wire bus
byte addresses[ADDR_SIZE * N_SENSOR];        // Here we store all the scanned addresses
#define ADDR(i) (&addresses[i * ADDR_SIZE]) // Macro to simplify our life
byte number_of_sensors;                     // Number of sensors found
word temperature[N_SENSOR];                 // Here we store temperatures

ZUNO_SETUP_SLEEPING_MODE(ZUNO_SLEEPING_MODE_FREQUENTLY_AWAKE);
// Define Z-Wave channels. Add more if you need more sensors
ZUNO_SETUP_CHANNELS(
        ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_TWO_BYTES, SENSOR_MULTILEVEL_PRECISION_TWO_DECIMALS, getterTemp1),
        ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_TWO_BYTES, SENSOR_MULTILEVEL_PRECISION_TWO_DECIMALS, getterTemp2)
);

void setup() {
  //Serial.begin(9600);
  number_of_sensors = ds18b20.findAllSensors(addresses);
}

void loop() {
  delay(2000);
  for (byte i = 0; i < number_of_sensors && i < N_SENSOR; i++) {
    // Read temperature
    temperature[i] = ds18b20.getTemperature(ADDR(i)) * 100;
    // Sending report
    zunoSendReport(i + 1); // Channels starts from 1
    //Serial.print(ADDR(i),HEX);
    //Serial.println(temperature[i]);
  }
  //delay(1000);
  zunoSetBeamCountWU(30);
  zunoSendDeviceToSleep();
}

```

**Compiler log:**
_Please paster the compiler output_
```
---------------------------------------------------------
                     FIRMWARE DATA                        
----------------------------------------------------------
	REVISION:02.15
	Z-WAVE FREQUENCY:EU
	ORIGINAL FW. CRC32: 66 C2 26 CC
	MAXIMUM CODE SIZE: 30 KB
	CURRENT  FW. CRC16: 4C A3
	RADIO CHANNELS: 02
----------------------------------------------------------
                     HARDWARE DATA                        
----------------------------------------------------------
	CALIBRATION
		MAIN:0a TX: 16 12
	DEVICE PUBLIC KEY:
		HEX: 1B 77 2B 10 A0 F5 8A 35 BA DC 44 FE 74 D6 BA 02 11 9C 58 77 89 6C CB 72 18 DF 38 B3 DC C9 12 0A
		DEC: 07031-11024-41205-35381-47836-17662-29910-47618-04508-22647-35180-52082-06367-14515-56521-04618
----------------------------------------------------------
                     PRODUCTION DATA                      
----------------------------------------------------------
	YOUR ZUNO S/N: 0F 12 06 B9 05 87 10 04 01
----------------------------------------------------------
	PRODUCTION TIME:
		WEEK:15 YEAR:2018
	WORKSTATION:
		NUMBER:16 SEQUENCE:1415 
	HARDWARE:
		CHIP:1721 REVISION:1025

```

**Debug output:**
_Connect to UART0, enable Logging->UART0, open logging monitor and dump the output. Should look like:_
```
not available
```

PoltoS, 2020-01-24:
@p0lyg0n1

silverman10, 2020-01-27:
Some more digging and i've noticed that this temperature is picked up from the below message payload

`Receive Message = 01 12 00 04 00 07 0A 60 0D 02 01 31 05 01 42 **4C FC** CA 00 87
`The above is from the controller.

Also seen that this is every 30 mins, which is the polling interval.

So, when the controller polls, the device, i get the below messages back.

```
Receive Message = 06
Receive Message = 01 04 01 13 01 E8
Receive Message = 01 05 00 13 B7 00 5E
Receive Message = 01 12 00 04 00 07 0A 60 0D 01 01 31 05 01 42 **70 AA** CA 00 EE

Receive Message = 06
Receive Message = 01 04 01 13 01 E8
Receive Message = 01 05 00 13 B8 00 51
Receive Message = 01 12 00 04 00 07 0A 60 0D 02 01 31 05 01 42 **4C FC** CA 00 87
```

Is this is specific response to the polling request, which the controller is mis-interpreting as a temperature reading or is it a temperature measurement?

PoltoS, 2020-01-27:
Would be interesting to see your getterTemp1. Looks that you multiply one more time the value by 10.

And yes, the controller is interpretting correctly the packet receied

silverman10, 2020-01-27:
the getterTemp1 is straightforward, it doesnt multiply the temp. it just returns temperature[1].
Also, if it was the getter, i would have expected all the temperature measurements to be off by the same factor, which isnt the case. I get the right readings, expect for the 30min poll intervals.

The other readings
```
2020-01-27 10:42:57.342 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 12 00 04 00 07 0A 60 0D 02 01 31 05 01 42 05 DC CA 00 EE
2020-01-27 10:44:58.802 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 12 00 04 00 07 0A 60 0D 00 01 31 05 01 42 05 E8 CA 00 D8
2020-01-27 10:44:58.832 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 12 00 04 00 07 0A 60 0D 01 01 31 05 01 42 05 E8 CA 00 D9
2020-01-27 10:44:59.648 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 12 00 04 00 07 0A 60 0D 02 01 31 05 01 42 05 DC CA 00 EE

```

PoltoS, 2020-01-27:
We will check it this week

silverman10, 2020-01-27:
> We will check it this week

thanks

silverman10, 2020-03-08:
Hi, any luck with this? I have done some more troubleshooting and this seems like a garbage value.

When i have size as 4 bytes and single precision, the values i get are 129772433 and 727477.2

p0lyg0n1, 2020-03-08:
Hi,

To avoid memory corruption use #define N_SENSOR 3 if you have 2 sensors on the bus.
always N_SENSOR  have to be  (n+1) where n is a maximum sensor count.
Try this code in loop.
 
`
void loop() {
  delay(1000);
  for (byte i = 0; i < number_of_sensors && i < N_SENSOR; i++) {
    // Read temperature
    temperature[i] = ds18b20.getTemperatureC100(ADDR(i)); // every value read gets about 1 second for ds18b20
    zunoSendReport(i + 1); // Channels starts from 1 // Only sets channel to report 
  }
  // It needs some time to report value 
  delay(100);
  zunoSetBeamCountWU(30);
  zunoSendDeviceToSleep();
}
`
Anyway I think that FLIRS sensor is a bad idea. The best way for sensors is fully sleeping devices. You can setup WakeUp interval for them using command class WEKEUP. ds18b20 is very slow for battery powered devices(it needs 1s for one measurement by default).

Manurev, 2023-10-04:
Hello, I'm a little new to this, I'm returning to this thread, with permission.
  Z1- ZWAVE V2 gen 7th with FIBARO's Zwave Yubii home HC1 as controller.
In my case, the Z_Wave code and transmission channels work well, but when the temperature reaches 125 degrees, higher temperatures of MORE THAN 125 DEGREES are no longer received and I don't understand why. The range I have checked is -29 to 125 degrees in case of "int temperature" variable. In case of "word or dword" dr 0 m to 125 degrees, it does not matter whether the size is two bytes or 4 bytes, the precision does not matter. There seems to be a limitation somewhere, but not in the code.
ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_FOUR_BYTES, SENSOR_MULTILEVEL_PRECISION_ONE_DECIMAL, GETTER)
Can anybody help me?
Greetings

PoltoS, 2023-10-04:
Please show your code

closed 2023-10-04 by PoltoS