Z-Wave-Me/Z-Uno-Core #122: sdcpp error

gerhil, 2020-11-26:
_Please fill your bug report here_

**More info:**
- **Z-Uno bootloader version:** 2.1.7 (release version or test-ucxx?)
- **Security:** none
- **Frequency:** EU
- **Device included:** yes
- **Device included securely by controller:** no
- **Controller:** Vera

**Connected peripherals:**
_Please add here a description, photo and/or schematics of your peripherals_

**Code:**
_If you can minimize your code to still reproduce the bug, please do it before reporting_
```C
your code here
```

**Compiler log:**
Found 1.8.x project structure (File:zuno_SimpleDimmer.ino.cpp). Converting it...
Error while detecting libraries included by C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389\sketch\zuno_SimpleDimmer.ino.cpp
Generating function prototypes...
"C:\\Users\\Gerald\\AppData\\Local\\Arduino15\\packages\\Z-Uno\\tools\\zuno_toolchain\\00.08.82/zuno_toolchain/compiler" arduino_preproc "C:\\Users\\Gerald\\AppData\\Local\\Temp\\arduino_build_307389\\sketch\\zuno_SimpleDimmer.ino.cpp" -r "C:\\Users\\Gerald\\AppData\\Local\\Arduino15\\packages\\Z-Uno\\hardware\\zw8051\\2.1.7" -DARDUINO_LIB_DISCOVERY_PHASE
C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389
Found 1.8.x project structure (File:zuno_SimpleDimmer.ino.cpp). Converting it...
           
"D:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Gerald\\AppData\\Local\\Temp\\arduino_build_307389\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\Gerald\\AppData\\Local\\Arduino15\\packages\\Z-Uno\\tools\\zuno_toolchain\\00.08.82/zuno_toolchain/compiler" build "C:\\Users\\Gerald\\AppData\\Local\\Temp\\arduino_build_307389/zuno_SimpleDimmer.ino" -r "C:\\Users\\Gerald\\AppData\\Local\\Arduino15\\packages\\Z-Uno\\hardware\\zw8051\\2.1.7" -i1_8 -idv 10813

	************* Building Arduino Sketch *************
	C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389/zuno_SimpleDimmer.ino
	***************************************************

	 --- USING a list of libraries from:
		C:\Users\Gerald\AppData\Local\Arduino15\packages\Z-Uno\hardware\zw8051\2.1.7\libraries
		E:\Users\Gerald\Documents\Arduino\libraries
	*** Collecting prototypes...
Preprocessing file: C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389\Custom.c with SDCPP... 

Preprocessing file: C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389\Stream.cpp with SDCPP... 
Compiling C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389\Stream_sdcpp_.cpp ...
Stream_sdcpp_.cpp:16:26:error:unknown type name 'uint32_t'
Stream_sdcpp_.cpp:19:25:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:22:31:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:22:42:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:24:26:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:38:5:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:39:5:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:43:23:error:expected class name
Stream_sdcpp_.cpp:53:13:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:54:13:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:65:13:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:68:27:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:69:13:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:69:30:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:75:18:error:unknown type name 'uint8_t'
Stream_sdcpp_.cpp:77:32:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:77:67:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:87:3:error:unknown type name 'size_t'
Stream_sdcpp_.cpp:87:35:error:unknown type name 'size_t'Unknown error:'NoneType' object has no attribute 'name' uCxx returned error code:-1

exit status 1
Error compiling for board Z-Wave>ME Z-Uno.

```
----------------------------------------------------------
                     FIRMWARE DATA                        
----------------------------------------------------------
	REVISION:02.15
	Z-WAVE FREQUENCY:RU

... (more lines up to)

	CHIP:1546 REVISION:4
```

**Debug output:**
_Connect to UART0, enable Logging->UART0, open logging monitor and dump the output. Should look like:_
```
0000:00 03
0000:00 00
0010:A0 00
0011:A0 00
0012:A0 00
```

PoltoS, 2020-11-28:
Please include your code

gerhil, 2020-11-28:
This is the code. This is a simple test program.

/* Based on SimpleDimmer example */

// LED pin number
#define LED_PIN 13

// Global variables.
byte lastDimmerLevel;
byte ledLevel = 0;

// Added because Vera only sees ZUNO_SWITCH_BINARY.
ZUNO_DISABLE(NO_MULTIBINARY_SWITCH0); 

// Set up the Z-Uno channels.
ZUNO_SETUP_CHANNELS(
  ZUNO_SWITCH_BINARY(getterLED, setterLED),
  ZUNO_SWITCH_MULTILEVEL(getter, setter)
  );

void setup() {  
  Serial.begin(9600);
  pinMode(LED_PIN, OUTPUT); //set up LED pin as output
  Serial1.println("Setup done!");
  //zunoSendReport(1);
  
}
void loop() {
  if (zunoIsChannelUpdated(1)) {
    // change relay state
    Serial.println("Channel changed! (1) = ");
    Serial.println(ledLevel);
  }
  if (zunoIsChannelUpdated(2)) {
    // change relay state
    Serial.print("Channel changed! (2) = ");
    Serial.println(lastDimmerLevel);
  }
}

void setter(byte value) {
  byte tempVariable;  
  if (value > 99) {
    // by Z-Wave specification, this value can't be more then 99
    value = 99;
  }
  // now we set the LED brightness
  analogWrite(PWM1, ((long)value) * 255 / 99);
  lastDimmerLevel = value;
}

byte getter(void) {
  return lastDimmerLevel;
}

void setterLED(byte value) {
  ledLevel = value;
}

byte getterLED() {
  return ledLevel;
}

gerhil, 2020-11-28:
Who you do the include twice in Custom.c?

-------------
lines #4-15

// Minimal includes

#include "ZUNO_legacy_channels.h"

// External varriables

// This file was automatically generated during the build...
// Do not modify it!

// Minimal includes

#include "ZUNO_legacy_channels.h"

gerhil, 2020-11-28:
For some reason "Print.h" in the temp build folder does not include:

p0lyg0n1, 2020-11-28:
Hi.

The code is ok and it was compiler on my PC the right way. So, I think the problem is in environment. 
I have checked the output of compiler and it looks strange. Do you have some extra libraries in E:\Users\Gerald\Documents\Arduino\libraries?
Can you zip all the build directory: "C:\Users\Gerald\AppData\Local\Temp\arduino_build_307389" and send it to me? I'll try to analyze its content. I haven't seen something like this before. 
Can you temporary move all the files from  E:\Users\Gerald\Documents\Arduino\libraries to another folder and try to build the sketch once more time?

	Yours,
                  Alex.

gerhil, 2020-11-28:
Hi Alex,

Thank you for your reply. I have disabled all compiler toolchains on my system and I've reinstalled Arduino IDE (1.8.13) and the z-uno plug in. Still the same problem. Then I discovered that the grbl library was the cause. It had a different Print.h file. I deleted the library and it worked :-)

Now I cannot associate my Fibaro Dimmer 2 to the z-uno with the example. What does it mean lifeline association of group #1 and setting 1:0? How do I do that on Vera?

Cheers,
Gerald.

p0lyg0n1, 2020-11-30:
Hi Gerald,
I haven't used Vera ever, but as I know it vas certified with Z-Wave Plus, isn't it?
So the Vera has to have interface for "multicnannel" associations. 
I didn't understand yet what you do want to do: control Fibaro Dimmer by Z-Uno or control Z-Uno by Fibaro dimmer.
Anyway there is a common approach and you have to setup associations on device from which you want to control another device. "LifeLine" group is only for reporting device status, it's a part of Z-Wave Plus standard. So, in case of Z-Uno you have to add extra group to control FibaroDimmer (I think the best choice is ZUNO_ASSOCIATION_GROUP_SET_VALUE_AND_DIM
 ).  
Please look at https://z-uno.z-wave.me/files/z-uno/QSG/Z-Uno-Quick-Start-Guide.pdf (page 24, "Association concepts") for more details. Good luck!

Best regards,
			Alex.

gerhil, 2020-12-11:
Hi Alex,

The Fibaro Dimmer needs to control the Z-uno. Any change of the dimmer should be read by the z-uno.
I think I have to associate dimmer group 3 to z-uno group 1 (see code above), Right?

Gerald.

The Dimmer 2 provides the association of five groups:

**1st Association Group „Lifeline”** reports state of the device. Main
Z-Wave Plus network controller should be added to this group. The
„Lifeline” group can handle only one device. It is not recommended
to modify this group.

**2nd Association Group „On/Off (S1)”** is assigned to switch no. 1.
Sends BASIC command class frame according to the state of the device.

**3rd Association Group „Dimmer (S1)”** is assigned to switch no. 1.
Sends MULTILEVEL SWITCH command class frame. Allows sending
dim/brighten command to associated devices.

**4th Association Group „On/Off (S2)”** is assigned to switch no. 2.
Sends BASIC command class frame according to the state of the device.

**5th Association Group „Dimmer (S2)”** is assigned to switch no. 2.
Sends MULTILEVEL SWITCH command class frame. Allows sending
dim/brighten command to associated devices.