Z-Wave-Me/Z-Uno-Core #124: Z1 V2 I CANNOT SEE TEMPERATURES HIGHER THAN 125 ºcº ON ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE CHANNEL

Manurev, 2023-10-04:
- Z-Uno version: 2
- Security: NONE and S0
- Frequency: EU
- Device included:** yes
- Device included securely by controller:myes
- Controller: Fibaro HC1 YUBII HOME
- ARDUINO IDE V.204
- I'm not a developer.
I try to send the values of a thermal solar panel up to 220 degrees.
I have tried everything. I have created several multilevel temperature channels, as the guide says. By default, and with double size and double precision celsius scale, with INT BYTE AND WORD variables. Source of variable "byte int word or float".
When I go from 125 degrees, or "12500" because the system seems to divide by 100. If I increase digits. The only thing that happens is that the decimals disappear, but I have never been able to transmit more temperature than 125 to the controller or the app. With or without decimals after the point.
When I increase to values above 125, I think there is some kind of oveflow. And this sends zero or stays at 125.
Is it a limitation of the dr fibaro driver? or am I configuring something wrong or my code is bad?
Multilevel channels are created well in the console. And read the data.
I know that in the z1 v2 guide it refers to signed values, but I don't quite understand how to translate that to send, for example, 210 degrees to the controller. And therefore to the app.
Oh, I haven't been able to send anything through the multilevel sensor either, it doesn't do anything, even if I change values.
I don't know what to do anymore, can someone help me?

Thank you very much in advance

// add library
#include "ZUNO_DHT.h"
// pin connection
#define DHTPIN 9 

DHT dht(DHTPIN, DHT22);

float humidity;      // here we will store the humidity
float temperature;   // here we will store the temperature
//word temperature0 = 5; // duplicado en el sistema pero corregido
dword temperature1 = 1347; //overflow en este modo con word o dwor para temp mayor que 125 parece que no puede haber float en la variable enviada por canal?
word temperature2 = 1498;// overflow en este modo que llega hasta 125 
word temperature3 = 102;// overflow en este modo que llega -29
//byte rele1 = 00;

//ZUNO_SENSOR_MULTILEVEL_TEMPERATURE(getterTemperature0),
//ZUNO_SENSOR_MULTILEVEL_WATER_TEMPERATURE(getterTemperature3),
// ZUNO_SENSOR_MULTILEVEL_TEMPERATURE(getterTemperature2),
// ZUNO_SENSOR_BINARY_GENERAL_PURPOSE(getterRele1)
// set up channels me falta un report
// getterTemperature1 he cambiado el metodo getter por variable, para eliminar la funcion geter por si hace algo con los valores
ZUNO_SETUP_CHANNELS(
      ZUNO_SENSOR_MULTILEVEL_TEMPERATURE(getterTemperature),
      ZUNO_SENSOR_MULTILEVEL_HUMIDITY(getterHumidity),
      ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_FOUR_BYTES, SENSOR_MULTILEVEL_PRECISION_ONE_DECIMAL,temperature1),
      ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_WATER_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_TWO_BYTES, SENSOR_MULTILEVEL_PRECISION_ONE_DECIMAL,temperature2),
      ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_SOIL_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_TWO_BYTES, SENSOR_MULTILEVEL_PRECISION_ZERO_DECIMALS,temperature3)
);

void setup() {
  dht.begin();
  Serial.begin(); 
  Serial.println("start");
}

void loop() {
  // obtaining readings from the sensor DHT
  humidity = dht.readHumidity();
  temperature = dht.readTemperature();
  Serial.print("Humidity = "); 
  Serial.print(humidity);
  Serial.print(" %  ");
  Serial.print("Temperature = "); 
  Serial.print(temperature);
  Serial.println(" *C");
  // send data to channels
  zunoSendReport(1);
  zunoSendReport(2);
  zunoSendReport(3);
  zunoSendReport(4);
  zunoSendReport(5);
  //zunoSendReport(6); 
  //zunoSendReport(7);
  // send every 30 second
  delay(30000);
}

byte getterTemperature() {
  return temperature;
}

byte getterHumidity() {
  return humidity;
}
/*
byte getterTemperature0() {
  return temperature0;
}
dword getterTemperature1() {
  return temperature1;
}

byte getterTemperature2() {
  return temperature2;
}
byte getterTemperature3() {
  return temperature3;
}

byte getterRele1() {
return rele1;
}
*/

PoltoS, 2023-10-04:
Please try to exclude and include it again. From your code, it looks like initially you had channels with one-byte values and then you changed to word/dword. But it is not allowed to change Z-Wave settings "on the fly", so Fibaro and Z-Uno remember the old size of the value. Please re-include and check it again.

Manurev, 2023-10-05:
Hello,
Thank you very much for the response, 
I have actually been careful about that, and whenever I have changed a channel or channel properties, such as size, scale, precision, or something in the getter function. I have successfully excluded, and then included, the Z1 module again. At least I have excluded and included about 20 times, one for each test. But it is true that your idea is interesting, because it seems that the changes are not recorded, somehow. Or at least it doesn't lead to the expected effect.

This code is a little more simplified than the one published previously, for testing. But, the result is the same.
In any case, I think that an int or word variable would be enough to represent 220 degrees. But it does not work. Without knowing why.
Kind regards

PoltoS, 2023-10-05:
Thanks for the simplified code. We will check it

Manurev, 2023-10-06:
Thanks PoltoS

Manurev, 2023-10-10:
I am sending you firmware data of my device.
Would it be convenient to activate clean NVM with the Arduino IDE?
before recording a Sketch on it?.
 ------------------------------------------------------------------------------------
                                        Z-Uno board information
          ------------------------------------------------------------------------------------
          
          FIRMWARE
          
          	 VERSION:		03.10
          	 BUILD_SEQUENCE:	00001842
          	 BUILD_DATETIME:	2023-01-17T16:38:12(MSK)
          	 SUPPORTED_HWREV:	0704
          
          LIMITS
          
          	 CODE:	178176 Bytes
          	 RAM:	16384 Bytes
          
          HARDWARE
          
          	 CHIP_UID:	 84 2E 14 FF FE 6A 28 10
          	 PROD_TIME:	 2021-07-07T20:33:01(MSK)
          	 PROD_SN:	 20
          	 LOCK:		 DBG_LOCKED
          
          LICENSE
          
          	 SUB_VENDOR:	0000
          	 BITMASK:	0000000000000000
          	 FEATURES:	[]
          	 CRC16:		84f9
          
          NETWORK
          
          	 HOMEID:	e3a1fcfe
          	 NODEID:	42

PoltoS, 2023-10-10:
Hi! We can't reproduce the issue. Are you able to turn on Zniffer (via PTI) on Z-Uno and check the traffic? Maybe the controller is wrong, not the Z-Uno.

Do you have another controller to check? Or do you have access to raw logs of your controller to see the exact bytes received from the Z-Uno

Manurev, 2023-10-11:
Hello, thank you for your response.
I don't know exactly how to activate Zniffer, I am not an advanced user, I am an Industrial Electronics Technician, and a little bit of programming and I have been learning Z-Wave for a while. More or less how I start.
I don't know how to activate Zniffer (via PTI) on Z1. But if you give me instructions I will do it. The controller is a YUBI Home De Fibaro HC1.
I don't have another driver to check, and I also don't know how to access the raw logs. Although I could do it with a tutorial.
What I would also like to know from you is, "if you have managed to transmit values greater than 125 degrees, for example 220 degrees, even through a variable, or getterfunction, in Z1 v2 Genth7. and other different controllers to the Yubii of fibaro HC1.?!
Because I wanted to know for sure, if it could be a problem with my controller, or a value limit in the z-wave channel protocol. (-29 to 125 degrees) or channel type, scala, or Z1.
ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_FOUR_BYTES, SENSOR_MULTILEVEL_PRECISION_ZERO_DECIMALS, GETTER or variable).
Or limitation in the command class!
If you give me instructions, and I can implement Znifer on the Z1, I will do it and do the necessary tests, but can it be done at the same time as running the Arduino program (Sketch)?
Maybe the channel was recorded with BYTE size the first time and has never been able to be modified again. I don't know. Although I have excluded all of them and left only one channel, for testing. And I have included again.
On the other hand, there is something in the Z1 guide that I don't understand well and I don't know if it is the problem.
He says like this:
Values returned by getter are signed. Values are interpreted according to the following transformation:
signed value = value / 10precision, for value < MAX_NUM/2
signed value = (value - MAX_NUM) / 10precision, for value ≥ MAX_NUM/2
Where MAX_NUM = 28*size and precision defined number of decimal digits after dot.
For example if size 1 and precision 0, value 25 represents 25,
for size 1 and precision 0, value 164 represents -92 = 164 - 256,
for size 1 and precision 1, value 25 represents 2.5 = 25 / 10,
for size 1 and precision 1, value 164 represents -9.2 = (164 - 256) / 10,
for size 2 and precision 1, value 366 represents 36.6,
for size 2 and precision 1, value 65535 represents -0.1 = (65535 - 65536) / 10,
for size 2 and precision 2, value 65535 represents -0.01 = (65535 - 65536) / 100.
????? I don't know how to interpret that.
Sorry for the inconvenience, and the length of my words, but I am excited about this project and solving all this and being able to contribute something.
Very grateful for the help you are offering me. Sincerely.

PoltoS, 2023-10-11:
The easier for you to:
- find how to see raw incoming packets in Yubii - not sure how, or
- apply for a Zniffer license in Z-Uno and listen for the packet sent out

For the Zniffer, you need to download the Zniffer app following the manual: https://z-wave.me/manual/zniffer/

Then apply for a Zniffer in PTI mode. We can provide it to you for free - just select wired transfer and let me know, so we activate it.

Then set up the Zniffer via PTI according to the manual. Once you have it, you can see what is sent to the network.

Manurev, 2023-10-12:
Hello PoltoS, thank you very much as always for the response.
I have some doubts, before starting with the Zniffer.
*First, very important!! Do you have proof, assured that the Z1 V2, genth7, can send values greater than 125 through the channel,???
ZUNO_SENSOR_MULTILEVEL(ZUNO_SENSOR_MULTILEVEL_TYPE_TEMPERATURE, SENSOR_MULTILEVEL_SCALE_CELSIUS, SENSOR_MULTILEVEL_SIZE_FOUR_BYTES, SENSOR_MULTILEVEL_PRECISION_ZERO_DECIMALS, GETTER or variable).??? In your systems, or tests?
For example with a Z1 and a Zway as a controller or other Z-Wave controller devices? To rule out this issue and clarify it.
*On the other hand, I have seen the Zniffer ART manual and I don't really know how to request the Zniffer for free, since I don't understand this phrase well;
(Then apply for a Zniffer in PTI mode. We can provide it to you for free - just select wired transfer and let me know, so we activate it.).
And could it work in PTI Mode, and/or Modem?
Anyway, if you can clarify how to request it, because I would like to try all this.
And thank you very much again for the collaboration.

PoltoS, 2023-10-12:
Hi

> First, very important!! Do you have proof, assured that the Z1 V2, genth7, can send values greater than 125 through the channel,???

For sure!

> Then apply for a Zniffer in PTI mode. We can provide it to you for free - just select wired transfer and let me know, so we activate it.

In the Zniffer app in [i]nfo window it will propose you to buy a license. And there you need to select the PTI option and press "Wired Transfer" under the PayPal link.

This Zniffer feature might be very helpful for your other projects. It is a Z-Wave sniffer tool

Manurev, 2023-10-13:
Hello PoltoS,
Well I have already installed the Zniffer/A.R.T software. user interface. And configure it. I didn't have to update my Z1 because it already has version 3.10. I got to the license payment site and chose transfer, like it said. Invoice for transfer. And it seems that something has been sent to them by email from mrodme@hotmail.com.
Waiting for your answer. All the best.

PoltoS, 2023-10-13:
Activated. Please go in the Zniffer UI and apply it

PoltoS, 2023-10-13:
Then you can see all packets sent to/from your Z-Uno and between other devices

Manurev, 2023-10-14:
Hello PoltoS
Yes,!!!!!!, I can now see the data packets, and I am trying to interpret them. I have already set high values to my temperature variables and I see that they appear in the tracer, I assume, that they are packets sent to the Z-Wave network. What I don't know are the values that are reaching the Fibaro YubII controller, in the preprocessing.
I'll keep trying, thank you.

PoltoS, 2023-10-14:
Paste here a screen of decrypted packets, please

Manurev, 2023-10-14:
[image]
[image]
[image]
[image]

Manurev, 2023-10-14:
Hello!
The last screen is a temperature value of -12 degrees, which appears in Two's Complement. If I'm not mistaken.
The truth is that the source and destination devices cannot be identified, I am not familiar with the fields, I do not know if they are in Hexadecimal. To try to understand the traces.
What is true is that they are the temperature sensors that I created in the Z1.
I have also saved an exported file in Json. In case you need it.
Again. Greetings

PoltoS, 2023-10-14:
All looks ok. The Z-Uno (# 43) is sending the correct data to the controller (# 1). If the same data is not recognized by Fibaro, then it is a Fibaro issue ;( you can try to change the size of the value to 1 or 2 bytes and pray Fibaro will recognize it correctly

Manurev, 2023-10-16:
Hi PoltoS
If you don't mind, I wanted to clarify some doubts,
- Size changes in the value of the Variables, and in the description of the channels, I already did it before, size, scale precision. 1 Byte, 2 Byte, 4 Byte. And it didn't work. When I send precision one or two, curiously, I can send up to 4 figures, a decimal after the period, but always the whole part of 3 figures and no more than 125.
- So there is a section in the Z1 guide, which refers to the values sent, when it is more than half of the maximum value, you have to send a different value that I cannot understand. Could this be the problem? Do I have to send transformed values?
It's here;

https://z-uno.z-wave.me/Reference/ZUNO_SENSOR_MULTILEVEL/

Values returned by getter are signed. Values are interpreted according to the following transformation:
signed value = value / 10precision, for value < MAX_NUM/2
signed value = (value - MAX_NUM) / 10precision, for value ≥ MAX_NUM/2
Where MAX_NUM = 28*size and precision defined number of decimal digits after dot.
For example if size 1 and precision 0, value 25 represents 25,
for size 1 and precision 0, value 164 represents -92 = 164 - 256,
for size 1 and precision 1, value 25 represents 2.5 = 25 / 10,
for size 1 and precision 1, value 164 represents -9.2 = (164 - 256) / 10,
for size 2 and precision 1, value 366 represents 36.6,
for size 2 and precision 1, value 65535 represents -0.1 = (65535 - 65536) / 10,
for size 2 and precision 2, value 65535 represents -0.01 = (65535 - 65536) / 100.
????
And finally, there are some functions that I don't understand well, and I wonder if I could change the properties of the Z-wave channels created and could they contribute something to the problem?

(Guithub)

//Sensor Multilevel Properties
#define SENSOR_MULTILEVEL_PROPERTIES_SIZE_MASK                                      0x07
#define SENSOR_MULTILEVEL_PROPERTIES_SCALE_SHIFT                                    0x03
#define SENSOR_MULTILEVEL_PROPERTIES_SCALE_MASK                                     0x18
#define SENSOR_MULTILEVEL_PROPERTIES_PRECISION_SHIFT                                0x05
#define SENSOR_MULTILEVEL_PROPERTIES_PRECISION_MASK                                 0xE0
#define SENSOR_MULTILEVEL_PROPERTIES_COMBINER(SCALE,SIZE,PRECISION) \
			(SIZE & SENSOR_MULTILEVEL_PROPERTIES_SIZE_MASK) | \
			((SCALE << SENSOR_MULTILEVEL_PROPERTIES_SCALE_SHIFT) & SENSOR_MULTILEVEL_PROPERTIES_SCALE_MASK) | \
			((PRECISION << SENSOR_MULTILEVEL_PROPERTIES_PRECISION_SHIFT) & SENSOR_MULTILEVEL_PROPERTIES_PRECISION_MASK)
??
Once with this information, I can address the problem with Fibaro support.
Thanks for the support.

PoltoS, 2023-10-20:
Hello!

Your examples of values are correct. So in your case, if you can't pass over 127, Fibaro is using 1 byte representation.

The template ZUNO_SENSOR_MULTILEVEL you are using should combine the right combination of flags to account for the selected size and precision.

PoltoS, 2025-08-26:
Hello! This is probably not the best thread to discuss this topic.

Each Z-Uno requires its own additional Zniffer/ART/PTI extension. You can swap the first one with the new one if you wish.

Manurev, 2025-08-27:
Yes, I'd like to change it to the new one, please, but I'm not sure how to do it.
And if we have to open a new thread, I'm not sure where to put it.
Thanks in advance.

Manurev, 2025-08-27:
The old one is already installed on a device with running software, I don't have it running as Zniffer, how could I acquire capacity for the new one?

PoltoS, 2025-08-27:
Please make an order, select wired payment and wait for us to approve it for you for free.

Manurev, 2025-08-27:
Which option do I choose, PTI or Advance radio tools?

Manurev, 2025-08-27:
Well, I have chosen PTI. Thank you very much.

PoltoS, 2025-08-30:
Yes, PTI shows all networks and all nodes

Manurev, 2025-09-01:
[Copia Trace Z-Uno 2 Temperature solar.json](https://github.com/user-attachments/files/22076432/Copia.Trace.Z-Uno.2.Temperature.solar.json)